RaspberryWallet/Backend #1: Push Button Module

Furfel, 2018-09-19:
**Prerequisites:**
* uses private key generated during first installation
* uses GPIO - pi4j
* timeout watchdog would be a good idea

**Testing:**
* GPIO can be mocked during testing
* private key can be generated before test and remain the same

How it works:
```
encrypt(data):
return AES.encrypt(PublicKey, data);
end

register():
end

check():
return GPIO.checkButton()
end

decrypt():
return AES.decrypt(PrivateKey, data);
end
```

stasbar, 2018-09-19:
What kind o `data` can hold button ?

Furfel, 2018-09-19:
https://github.com/RaspberryWallet/Backend/wiki/How-Modules-work

PatrykMilewski, 2018-10-26:
@pfoof @stasbar I see this in code, but is this fully implemented?

PatrykMilewski, 2018-11-02:
@pfoof can you finish it today?

It's funny, that it's #1 issue :1st_place_medal:

PatrykMilewski, 2018-11-07:
Weekly reminder to finish this branch @pfoof :hourglass_flowing_sand:

PatrykMilewski, 2018-11-12:
Does it even make sense to keep this branch, since there are like one thousand conflicts with current master?

@pfoof @stasbar

stasbar, 2018-11-12:
I have no idea why is this issue/branch still open.

PatrykMilewski, 2018-11-13:
Is Push Button module fully implemented? I think, it is not
https://github.com/RaspberryWallet/Backend/blob/master/Manager/src/main/java/io/raspberrywallet/manager/modules/pushbutton/PushButtonModule.java

stasbar, 2018-11-13:
I think it is. Button cannot hold any cipher, so it can just return the data. Or we can pass it throught NOT function

stasbar, 2018-11-13:
And we must test it in practice. @pfoof

PatrykMilewski, 2018-11-14:
It should test if GPIO is in correct state and then return data, if it is not then throw exception, i would do it as simple as that

saving static encryption key to file would give us nothing imo

PatrykMilewski, 2018-11-16:
Push Button module was implemented in other branches. Closing

closed 2018-11-16 by PatrykMilewski