RaspberryWallet/Backend #3: Increase simplicity by merging ServerHttp and Manager build system

stasbar, 2018-09-19:
Since Maven and Gradle modules are incompatible with each other, current modularity scheme is highly problematic. Maven can include dependency only on resulting artifact in .jar or in maven repository. I suggest changing both modules into either Maven or Gradle build system. I suggest using Gradle, how about you?

Furfel, 2018-09-20:
When importing to IJ these are two separate projects, are we planning to use CI? If so, can't it actually take two different paths and compile?

stasbar, 2018-09-20:
After discussion we decided to not use CI. Two build system in one project brings more problems than it's needed. Especially when one module is used only in the latter one. The problem is, when you change code in Server, then Manager is not aware of this change since it depends on Maven repo artifact nor on Server source code. Instead of hacking and tweeting this two incompatible modules, I'm more likely to migrate Server to Maven build system. (But prefer migrate Manager to Gradle since I'm more familiar with it)

PatrykMilewski, 2018-09-21:
I vote for Maven, since we are going to work together on Manager and even if ServerHttp needs kotlin, it's still very small module, comparing to Manager.

I'm going to fix pom.xml for Manager after we agreed in #8 to keep with Java. It's going to be short and clean.

closed 2018-09-22 by Furfel