RaspberryWallet/Backend #5: Add Module timeout for CheckRunnable

Furfel, 2018-09-20:
Create an additional variable which will stop checking if interaction with the module (`check()`) was done after a specific period of time.

**Pseudocode current:**
```
boolean run
while(run) {
    if(check())
         process()
}
```

**Proposed:**
```
boolean run
int startTime = CurrentTime()
int timeout = 3000
while(run) {
    if(check())
        process()
    if(CurrentTime() - startTime > timeout){
        run = false
        status = STATUS_TIMEOUT
    }
}
```

stasbar, 2018-09-20:
Why would it ever execute process() two times ?
Shouldn't it be just 
 ```
boolean run
while(run) {
    if(check()){
         process()
         run = false
    }
}
```

stasbar, 2018-09-20:
and btw.
```
boolean run
while(run) {
   if(check()){
        process()
        run = false
   }
    Thread.sleep(1000)
}
```
Maybe less overhead

Furfel, 2018-09-21:
That's a bug xd

Furfel, 2018-09-21:
Committed change. However, `Thread.sleep` requires `InterruptedException` to be caught.

closed 2018-09-22 by Furfel