RaspberryWallet/Backend #6: Blakley and Shamir shared secret algorithms are orderless

stasbar, 2018-09-20:
As far as I know, both Blakley and Shamir shared secret algorithms are orderless.
So I think that 

```java
public class KeyPartEntity {
        ...
	public int order;
        ...
}
```

is useless in this case. Or maybe it's different "order"? Comments in code, are highly welcome.

PatrykMilewski, 2018-09-20:
There is link to this class file:
https://github.com/RaspberryWallet/Backend/blob/master/Manager/src/main/java/io/raspberrywallet/manager/database/KeyPartEntity.java

Yes, it is useless. 

Actually I think this class needs some refactor, since there are different level of abstractions in the same method, which makes it hard to read. Also Lombok annotations should be outside of the class and method name "clean" should be "destroy" and maybe we should think about making destroyable interface, which will be implemented by this. It will cause less dependencies in code, since in many situations there will be only single call:
`key.destroy();`

stasbar, 2018-09-20:
If `order` is the indicator of key portion to decrypt, then I don't think that just `order` as an `index` is sufficient in our case.

PatrykMilewski, 2018-10-07:
is it fixed?

PatrykMilewski, 2018-10-10:
after meeting we agreed to call this:

**order**

closed 2018-10-10 by PatrykMilewski