RaspberryWallet/Backend #13: PIN Module

Furfel, 2018-09-23:
```
String pin = null;
encrypt(data):
return AES.encrypt(PublicKey, data);
end

register():
pin = null;
setStatus("Enter PIN code. <form:pin>");
end

check():
return pin != null && pin.length > 3
end

decrypt():
return AES.decrypt( PrivateKey(pin), data );
end
```
What's more, frontend should react on `<form:pin>` and show the form, which will post data to Module.
Needs extending Module.java to hold and be able to handle dynamic input: `HashMap<String, String>`

PatrykMilewski, 2018-09-27:
What about entering PIN with physical keyboard on RPI? Are we going to do this only by web panel?

Furfel, 2018-09-29:
@PatrykMilewski that would be a different module

PatrykMilewski, 2018-10-07:
> @PatrykMilewski that would be a different module

ok, you are right

PatrykMilewski, 2018-10-26:
@pfoof @stasbar  I see this in code, but is this fully implemented?

stasbar, 2018-10-26:
Yes, but I used xor instead of aes :D

PatrykMilewski, 2018-10-26:
Okay, then please make a new issue nad assign me, i will drive it to the
end :D

PR #65 (2018-11-02): Implemented AES encryption for PIN module
Implemented AES encryption for PIN module
Added tests for PIN module

This closes #13 

Took 1 hour 7 minutes
closed 2018-11-02 by PatrykMilewski