RaspberryWallet/Backend #19: Missing ServerHttp package on JitPack repository

PatrykMilewski, 2018-09-26:
After I tried to compile Manager module, it failed with this message:

```
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 0.468 s
[INFO] Finished at: 2018-09-26T18:40:07+02:00
[INFO] Final Memory: 9M/241M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal on project Manager: Could not resolve dependencies for project io.raspberrywallet:Manager:jar:1.0: Failure to find io.raspberrywallet:ServerHttp:jar:1.0 in https://jitpack.io was cached in the local repository, resolution will not be reattempted until the update interval of jitpack.io has elapsed or updates are forced -> [Help 1]
```

I think that it was present in local maven repository, but not published to JitPack repo.

PatrykMilewski, 2018-09-26:
Actually it works if I call compile on root pom.xml, but I'm not sure if it's expected behavior.

stasbar, 2018-09-26:
Well, it's also strange for me, that Maven doesn't have separation for local and remote dependencies.

PatrykMilewski, 2018-09-26:
@stasbar it does have local repo for dependencies and remote is in our case jitpack + maven central
after compiling it with root pom.xml, everything is fine, but still there should be 1.0 version on jitpack

stasbar, 2018-09-26:
I meant that in pom.xml you can not differentiate "use local module dependency" or "use remote module dependency" it looks like it will try to use local, and if it's absent then it will search in remote repos. 

On jitpack is uploaded only 0.1.0-alpha version

PatrykMilewski, 2018-09-27:
Can we just change out version in pom to match there in JitPack? It would solve the problem

stasbar, 2018-09-27:
But the whole point is to be independent from jitpack, we don't need it anymore. Sever and Manager are too tightly coupled to separate and distribute them on public maven repositories.

PatrykMilewski, 2018-09-27:
Okay, then can we get rid of JitPack? Do we need changes in code?

stasbar, 2018-09-28:
No, everything is set. We can get rid of jitpack repo (and easy-logger)

closed 2018-10-04 by PatrykMilewski