RaspberryWallet/Backend #29: Raspberry Performance

stasbar, 2018-10-01:
We need to find the best software stack that will handle our requirements.
**OS**: We should try both Rasbian and DietPi. It may turn out that Raspbian is more stable and so faster.
**JDK**: According to https://raspberrypi.stackexchange.com/a/43912 We should use Oracle JDK
**ServerFramework**: We should try VertX, Ktor(Netty) and Ktor(Jetty) and find which perform best.

pfoof, 2018-10-03:
Benchmark results:
https://docs.google.com/spreadsheets/d/1a1OVP222QlkPdzlxxynXhlwhVE-9O08XIMxoJKJLtJc/edit?usp=sharing

stasbar, 2018-10-03:
Nice. So indeed, Oracle is slightly better.

pfoof, 2018-10-03:
`sudo ln -s /usr/lib/jvm/jdk1.8.0_181/bin/java /etc/alternatives/java`

stasbar, 2018-10-03:
Sorry @pfoof didn't get the joke :(

pfoof, 2018-10-04:
It's not a joke, it's a command to pin a different version of java

PatrykMilewski, 2018-10-07:
So indeed, we stay with Oracle and DietPi?

PatrykMilewski, 2018-10-10:
> So indeed, we stay with Oracle and DietPi?

meeting: yup

closed 2018-10-10 by PatrykMilewski