RaspberryWallet/Backend #38: SMS Module

PatrykMilewski, 2018-10-10:
Use external service for sending SMS and let user enter his own API key. 

We have to choose service provider with simple REST API

PatrykMilewski, 2018-11-12:
Do we really want to implement it? I was thinking about starting this after closing #84, but I'm not 100% sure.

If not, can you give me something, that is independent from other parts of system like modules? I just don't want to waste time for asking stasbar, how something works

@stasbar @pfoof  What do you guys think?

pfoof, 2018-11-13:
Leave it as nice to have, it would look more professional and make us seem more busy :^)

PatrykMilewski, 2018-11-13:
I think that I will jump straight to Heroku / Documentation then

stasbar, 2018-11-13:
Ye, we have enough modules for demo.