RaspberryWallet/Backend #39: Wallet configurator

PatrykMilewski, 2018-10-10:
For now we are not going to creat graphical wallet configurator.

We will use yaml format for creating it.

stasbar, 2018-10-12:
What kind of configuration should be in it ?

PatrykMilewski, 2018-10-26:
@stasbar it should look like Docker-Compose yaml, atleast for modules:

```

```version: '0.1'

modules:

  pin:
    max-retry: 5
    instances: 3      # parallel modules

  authorization-server:
    hostname: 89.89.89.89
    port: 8080
    endpoints:
      set-secret: /authorization/secret/set
      overwritte-secret: /authorization/secret/overwritte
    https: true
    
  button:
    some: configuration
    
wallet:
  autolock-time: 1800
  etc: etc

stasbar, 2018-10-26:
There must be some standardized way of naming the modules in configuration file. How should the name `pin` be linked with PinModule other way than hardcoding.

PatrykMilewski, 2018-10-26:
Maybe let's use get name of module method that all modules have to
implement, and it would be forced to use - separators

commit cc212f42 (2018-11-01): merge with #64
closes #39
dependency injection into Bitcoin and Database
fix tests

Took 34 minutes