RaspberryWallet/Backend #47: Auto locking manager

stasbar, 2018-10-14:
I suggest to implement auto locking mechanism, that will lock the device, after some period of time ( (default: 10min ?)

It would be as easy as
```java
class Manager{

    private int autoLockRemainingMinutes = 10;

    Manager(){
        Timer timer = new Timer();
        timer.scheduleAtFixedRate(new TimerTask() {
            @Override
            public void run() {
                if (autoLockRemainingMinutes == 1) {
                    lockWallet();
                    timer.cancel();
                }
                --autoLockRemainingMinutes;

            }
        }, 60000, 60000);
    }
    
    void tap(){
        autoLockRemainingMinutes = 10;
    }
}
```

stasbar, 2018-11-07:
It's working fine in practise

stasbar, 2018-11-07:
It's implemented in 50%, because It's working in backend, but there is no frontend view, like timer or just TextView to display time remaning. I will implement it with WebSockets

PatrykMilewski, 2018-11-17:
Checking state, @stasbar how it looks right now? Same as 10 days ago?

commit f5dbd61f (2018-11-22): updated webapp
closes #47

Took 50 seconds
PR #106 (2018-11-24): Autolock
fixed Autolock websocket
changed default autolock to 5mins
updated WebApp where progress bar indicates autolock time remaining 
updated README with gif demo
closes #47 

Took 20 minutes 50 seconds
closed 2018-11-24 by PatrykMilewski