RaspberryWallet/Backend #52: Encrypted Database for keyparts

PatrykMilewski, 2018-10-26:
This will be blocker, it's important to implement this. Any ideas which database we should use?

Task for @pfoof

stasbar, 2018-10-26:
As far as i remember, we decided to use just aes-encrypted file serialized with JSON

PatrykMilewski, 2018-10-26:
Okay, then we have to make logic for it

PatrykMilewski, 2018-10-27:
@stasbar are we going to use Java built-in serialization, or you want to use something else?

pfoof, 2018-10-27:
@PatrykMilewski we already have serialization using jackson. `io.raspberrywallet.manager.database.Database` keeps a `WalletEntity` which holds an array of `KeyPartEntity`.

PatrykMilewski, 2018-10-27:
@PatrykMilewski will implement AES encyrption in Database class

stasbar, 2018-10-30:
Since keyparts are native encrypted, do we need to encrypt the database ?
By native, I mean that keypart is never ever saved in raw/decrypted form. 
It might bring some complexity, like a password to encrypt database file.

pfoof, 2018-10-30:
Structure of the database should also be protected.

PatrykMilewski, 2018-10-30:
Adding something to what @pfoof said, it's good idea to encrypt whole database, since some modules might encrypt keypart in naive way, like xor operation or something, they still may be protected when somebody might try to attack wallet

stasbar, 2018-10-30:
You are right, let's encrypt it as well

PR #64 (2018-11-01): Wallet database encryption
This closes #52 

CI is not passing due to problems with CircleCI
closed 2018-11-01 by stasbar