RaspberryWallet/Backend #59: Database file location

PatrykMilewski, 2018-10-29:
Database is encrypted then serialized and saved to file as byte array in file database.bin

Default path is `./database.bin` which probably project root path, or something like that.

I suggest to move it to /opt/wallet, but this change needs discussion, since it's important, where we store most important data.

@pfoof @stasbar

stasbar, 2018-10-29:
I think that it's must have change, because I can't imagine the situation where the cwd execution  decide where the database is saved. 
I will do the same with blockchain(`.spvchain`) and `.wallet` files. They will be saved into `/opt/wallet/bitcoinj`

stasbar, 2018-10-30:
Confirmed 😄

pfoof, 2018-10-31:
The application is executed in /opt/wallet which is the `~` directory for its user.

stasbar, 2018-11-01:
@pfoof Well, that changes a lot.
So, should the base directory be defined by enviroment variable $HOME or in config.yaml file ? 
Or maybe $HOME will be default, and config.yaml will be able to overwrite it ?
@pfoof @PatrykMilewski

PatrykMilewski, 2018-11-02:
Can we make a decision and close this issue? The original question was already answered and implemented.

PatrykMilewski, 2018-11-07:
Keep with full paths