RaspberryWallet/Backend #60: Implement custom logic wallet locking/unlocking

stasbar, 2018-10-29:
After consultation with BitcoinJ team, it turned out that our (de/en)cryption modules case, doesn't fit to  the library out-of-the-box. There must be some wrapper logic involved, to manage Peers, Blockchain synchronization and wallet (de/en)cryption.

PatrykMilewski, 2018-10-30:
Could you explain, why we cannot use the original idea? (splitting encryption key used by BitcoinJ)

stasbar, 2018-10-30:
Yes, we can, and we are. But it turned out that bitcoinJ by default doesn't allow to save wallet data on demand. We must enforce that wallet data never gets saved in raw decrypted form. BitcoinJ by default save the wallet in decrypted form many times during synchronization. I have already found a way to do that, it's not so hard as I thought, but might be tricky.

PatrykMilewski, 2018-10-30:
Okay, I understand, I hope it will go smooth

PatrykMilewski, 2018-11-03:
@stasbar I was thinking about different approach to achieve this.

Can we specify the path, where BitcoinJ is writting it's files? If yes, we could do something like this:
https://unix.stackexchange.com/questions/188536/how-to-make-a-temporary-file-in-ram

Directory given to BitcoinJ would be placed on virtual partition, which would be in RAM. We would achieve something like "stateless" wallet.

stasbar, 2018-11-04:
Well, it's not stupid idea, but a bit too late. This implementation is almost done. I just need to tweak UI, write more tests and it's good to go. It will also give us way more flexibility.

PR #70 (2018-11-07): BitcoinJ custom logic
closes #60 
Custom bitcoin logic implementation
Window for huge space for optimization (delayed for "nice to have feature")
New UI in Webapp

Took 7 hours 59 minutes
closed 2018-11-07 by stasbar