RaspberryWallet/Backend #61: Parallel modules

PatrykMilewski, 2018-10-30:
I was thinking about easy solution, how to implement parallel and nasted modules. I couldn't eat.. I couldn't sleep..

Then I've got an idea! Let's just implement module, that takes list of modules in constructor. Then for encrypt and decrypt operations, it needs user input, where user specifies module that he wants to unlock. Take a look:
```
ParallelModule(List<Module> modules) {
     this.modules = modules;
}

public byte[] encrypt(byte[] payload) {
     int moduleNumber = userInput.moduleNumber;

     modules.set.userInput(userInput.moduleInput(moduleNumber));
     return modules.get(moduleNumber).encrypt(payload);
}
```

We can nest them like that:
```
List<Module> modules = new ArrayList<>();
List<Module> nastedModules = new ArrayList<>();
modules.add(new ParallelModule(nastedModules));

Module rootModule = new ParallelModule(modules);
```

What do you think?

stasbar, 2018-10-30:
It may work, but it's only the tip of the iceberg. 
We would also need to also handle this case in webapp and probably in database structure. It would be really cool to implement this feature, but do we have time ? Right now I'm quite busy with BitcoinJ puzzles and front-backend communication.

PatrykMilewski, 2018-10-30:
yup, I will create a new milestone called "After release features" and add this