RaspberryWallet/Backend #66: Java Serialization in cryptography

stasbar, 2018-11-02:
Could you please exaplain why do we need to involve Java Serialization to our cryptography ?
For example:
```java
    @Override
public byte[] encrypt(byte[] payload) throws RequiredInputNotFound, EncryptionException {
        Password password = new Password(getPin());
    
        AESEncryptedObject<ByteWrapper> encryptedObject =
                CryptoObject.encrypt(new ByteWrapper(payload), password);
        
        return SerializationUtils.serialize(encryptedObject);
```
I don't get why the API looks like this, and not just 
```java
    @Override
public byte[] encrypt(byte[] payload) throws RequiredInputNotFound, EncryptionException {
        Password password = new Password(getPin());
        return CryptoObject.encrypt(new ByteWrapper(payload), password);
```

PatrykMilewski, 2018-11-02:
Well, we can use `CryptoStream` instead, but using `CryptoObject` is shorter.

It's simple idea:
1. You wrap your data in any wrapper, or you can just pass Object, that implements Serializable, so it would be simpler.
2. Now you can simply encrypt your object and in return, you get wrapped encrypted bytes with `AESEncryptedObject` class, that holds all the info about used parameters for AES.
3. Since we have to return byte array, then we can just Serialize `AESEncryptedObject` wrapper and return this.

I see the problem, that it uses Serialization, which is generally slow and unwanted. The better way would be to use `CryptoStream` instead, but to be honest, tests of these were lost somewhere and I couldn't get it to work and then used `CryptoObject`.

PatrykMilewski, 2018-11-02:
The API looks like this, because it was took from BSK project, where AESEncryptedObjects was stored in collection, so there was no point in returning byte array.

stasbar, 2018-11-03:
Ok, thanks. I get it 👍

closed 2018-11-03 by stasbar