RaspberryWallet/Backend #76: Wallet operations history

stasbar, 2018-11-09:
Implement operations history

PatrykMilewski, 2018-11-11:
Can we discuss about how it will be stored? Storing logs in JSON is the best approach atm in my opinion. Reading them with other tools like elasticsearch, would be then really easier (not saying that we are going to do this, just our users)

PatrykMilewski, 2018-11-11:
Also we may think about adding a layer of abstraction to logging in overall. Then if we would like to improve our logging system, then it would be a lot easier to modify our custom logging interface

PatrykMilewski, 2018-11-17:
@stasbar So basically operations history is stored in file, but theere is no frontend logic to show it?

If yes, then this Issue should be renamed to "Displaying wallet operations history" instead, or something like that

stasbar, 2018-11-17:
Right