RaspberryWallet/Backend #81: Random string generator

PatrykMilewski, 2018-11-12:
When I was debugging something, I've found that my random string generator was not really that random:
```
byte[] randomBytes = new byte[PASSWORD_SIZE_IN_BYTES];
random.nextBytes(randomBytes);
return new String(randomBytes, Charset.forName("UTF-8"));
```
I've found it in Google and I've been using it for some time. The problem is, that `random.nextBytes` may generate negative numbers, which will be converted to character with code 65533. In a result, half of a random string is going to be 65533 chars, since getting a nagative character is around 50% chance. 

A lot of google answers about "how to generate random string in java" is affected with this. For example:
https://www.baeldung.com/java-random-string

I will fix it on my actual feature branch (authserver-https). I'm writting this to just warn you, because you may used it like this somewhere else and without test special for this, it's really hard to find it.

stasbar, 2018-11-12:
Good point !

stasbar, 2018-11-13:
Isn't it simpler to just use Base64/58/32 encoding?

PatrykMilewski, 2018-11-13:
Yes, it is simpler, I will change that in #84

PR #84 (2018-11-14): AuthorizationServer module HTTPS support
This closes #77 
Please don't merge until #83 is resolved

- Added HTTPS support for authorization server module
- Refactor of Apache Http client wrappers
- Refactor of Authorization Server Module Tests
- Fixed bug in AuthorizationServerAPI, where isRegistered flag was set incorrectly, after turning on the wallet again, by removing isRegistered and isLoggedIn flags, which will cause small performance decrease
- Removed https flag from AuthrizationServerConfig
- Added Lombok Getter annotation to AuthorizationServerConfig class
- Changed Base64 encoding/d [... cut]
commit 44f96120 (2018-11-14): Small SecureAppacheHttpClient refactor

Fixed bug in AuthorizationServerAPI, where isRegistered flag was set incorrectly, after turning on the wallet again, by removing isRegistered and isLoggedIn flags

Fixed #81 where incorrect random strings were generated

Removed https flag from AuthrizationServerConfig

Added Lombok Getter annotation to AuthorizationServerConfig class

Added acceptUnstrustedCerts flag to AuthorizationServerConfig

Changed Base64 encoding/decoding to url safe Base64

Removed unused Base64Secret class

Took 4 hours 27 minutes

Took 4 minutes
commit 540b180f (2018-11-15): Small SecureAppacheHttpClient refactor

Fixed bug in AuthorizationServerAPI, where isRegistered flag was set incorrectly, after turning on the wallet again, by removing isRegistered and isLoggedIn flags

Fixed #81 where incorrect random strings were generated

Removed https flag from AuthrizationServerConfig

Added Lombok Getter annotation to AuthorizationServerConfig class

Added acceptUnstrustedCerts flag to AuthorizationServerConfig

Changed Base64 encoding/decoding to url safe Base64

Removed unused Base64Secret class

Took 4 hours 27 minutes

Took 4 minutes
closed 2018-11-14 by PatrykMilewski