RaspberryWallet/Backend #86: First release - finish main features

PatrykMilewski, 2018-11-15:
We have to finish implementation phase. 

Let's list what we got left:
- confirm that PINModule is finished (#1) @pfoof 
- refactor Module interface (remove check, move reposibility to enc / dec)
- ???

@pfoof @stasbar

PatrykMilewski, 2018-11-15:
I think that we have to implement 2. due to problems with some modules, that currently may not work with / without check call

PatrykMilewski, 2018-11-15:
> We need to test everything in practice on rpiz @pfoof we don't even know if it will work.
> By everything I mean, scenario when we use all modules to lock wallet, synchronize and send funds.

Yup but right now I'm 100% sure, that some of modules will not work due to 2.

stasbar, 2018-11-15:
Ok, so who is willing to implement 2. ?

PatrykMilewski, 2018-11-15:
I can, but somebody have to verify Modules Manager logic if it works correctly then

PatrykMilewski, 2018-11-15:
Created #87 for 2.

stasbar, 2018-11-15:
PR for 2. https://github.com/RaspberryWallet/Backend/pull/88

PatrykMilewski, 2018-11-16:
We need tests 3. @pfoof

PatrykMilewski, 2018-11-17:
Closing this issue, 3. will be continued in #98

closed 2018-11-17 by PatrykMilewski