RaspberryWallet/Backend #87: Refactor of Module abstract class

PatrykMilewski, 2018-11-15:
Due to missunerstandings of official Module API provided in this class, there are some problems with current Modules implementations.

We have to refactor this class to work like this:
1. Set needed inputs.
2. Call enc / dec method
3a. If the requirements for this operations didn't meet, then exception is thrown and the error message is passed to user
3b. Method returns correct output.

What do you guys think? I think, that it should be as simple as possible.

@stasbar @pfoof

PatrykMilewski, 2018-11-15:
Branch for this issue:
https://github.com/RaspberryWallet/Backend/tree/manager-refactor

stasbar, 2018-11-15:
Isn't it as simple as remove the `check()` and add input check to enc/dec ?
in ExampleModule, PinModule and PushButtonModule this method body is stateless, it just return if the input is set. We can move this checks to enc/dec and throw RequiredInputNotFound if it's not set.

stasbar, 2018-11-15:
Of we can add validateInput() where the validation needs to be done and throw `RequiredInputNotFound` if its not present

PatrykMilewski, 2018-11-15:
As you said first, remove check and do check work in enc dec, I started refactoring it already like this

PR #88 (2018-11-16): Module Class Refactor
replace `check()` with `validateInput()` which must throw `RequiredInputNotFound` if the input is not present or incorrect.
add public wrapper methods `validateAndEncrypt()` which will force framework to `validateInputs()` before en/decryption
modules override protected methods decrypt/encrypt
implemented changes in `PinModule`, `ButtonModule` and `ExampleModule`

@PatrykMilewski please approve this concept and implement those changes in `AuthorizationServerModule`

Closes #87 

server port as config property
default port 80
default secure-port: 433

Took 1 hour 59 minute [... cut]
closed 2018-11-16 by PatrykMilewski