RaspberryWallet/Backend #107: Abstract Module class is too complicated

PatrykMilewski, 2018-11-23:
I'm writting README about how modules works and I found, that if you look into a code of a Module class:
https://github.com/RaspberryWallet/Backend/blob/master/Manager/src/main/java/io/raspberrywallet/manager/modules/Module.java

Then you notice, it's really compilcated and long. It's kinda problem, due to users getting demotivated to implement their custom modules. I belive, that user should see just abstract methods, that MUST be implemented, for module to work with our wallet. Also I think, that Module class should be splitted to 2 classes.

My proposal of solution, waiting for yours, because I don't really like it:
- ParentModule class: contains all the abstract methods, all protected methods and default logic for all of them
- Module class: extends after ParentModule, contains again abstract methods but just them, nothing more

Or maybe something with interfaces?

stasbar, 2018-11-24:
I agree that this class should be refactored. But I don't think it's too long. It's just 220 lines, and `Module` class is the core functionality of our project, so it naturally becomes big.
The interface approach may work. I will try to implement it.

In addition to that, the Module class is responsible for 4 functionality domains:
* Inputs - Basically CRUD of inputs map
* (En/De)cryption of key part.
* Configuration - Holding type, and pulling config from configuration tree
* View State - Holding the status string, and UiHtmlForm

I think it may be refactored as well, but I'm not sure if it will bring any profit.

stasbar, 2018-11-24:
And what you suggest is
```java
public abstract class SecurityModule<Config extends ModuleConfig> extends Module<Config> {
    public SecurityModule(@NotNull String initialStatusString, Class<Config> configClass) throws IllegalAccessException, InstantiationException {
        super(initialStatusString, configClass);
    }

    public SecurityModule(@NotNull String initialStatusString, Configuration.ModulesConfiguration modulesConfiguration, Class<Config> configClass) throws IllegalAccessException, InstantiationException {
        super(initialStatusString, modulesConfiguration, configClass);
    }

    protected abstract String getDecription();

    @Nullable public abstract String getHtmlUi();

    protected abstract void validateInputs() throws RequiredInputNotFound;

    protected abstract byte[] encrypt(byte[] keyPart) throws EncryptionException, InternalModuleException;

    protected abstract byte[] decrypt(byte[] payload) throws DecryptionException, InternalModuleException;
}
```
And then the `CustomModule` needs to extend this class, right ?

PatrykMilewski, 2018-11-24:
Well, you are right, that it contains a lot of logic, but I think, that methods are blured by default implementation. 

Maybe let's forgive about more inheritance and just add Interface, that is going to make it clear, what public methods does Module implements?

stasbar, 2018-11-24:
Ok, so what you suggest is:
```java
public abstract class Module<Config extends ModuleConfig> implements IModule
```
and IModule
```java
public interface IModule {
    /**
     * Used in all sort of identifications like config.yaml, internal module mapping and UI naming.
     * For now, it's just simplified SimpleClassName
     *
     * @return module identifier.
     */
    String getId();

    /**
     * @return Server-side module representation
     */
    io.raspberrywallet.contract.module.Module asServerModule();

    /*
     * (De/En)cryption
     */

    /**
     * this wrapper enforce module to validateInputs and throw exception if they are absent
     *
     * @param keyPart - unencrypted key part
     * @return encrypted payload
     */
    byte[] encryptKeyPart(byte[] keyPart) throws EncryptionException, RequiredInputNotFound, InternalModuleException;

    /**
     * this wrapper enforce module to validateInputs and throw exception if they are absent
     *
     * @param payload - encrypted payload
     * @return decrypted key part
     */
    byte[] decryptKeyPart(byte[] payload) throws DecryptionException, RequiredInputNotFound, InternalModuleException;

    /*
     * View State
     */

    String getDescription();

    /**
     * this function should return HTML UI form or null if not required
     */
    @Nullable
    String getHtmlUi();

    /**
     * Returns status of the module to show to the user
     *
     * @return message
     */
    @NotNull
    String getStatusString();

    /**
     * Setting the status message for the user
     *
     * @param status - new status
     */
    void setStatusString(@NotNull String status);

    /*
      Inputs CRUD
     */

    /**
     * Gets the value which user has submitted
     *
     * @param key - parameter key
     * @return - value of the parameter
     */
    @Nullable
    String getInput(String key);

    /**
     * Sets input for this Module from user
     *
     * @param key   - key of the parameter
     * @param value - value of the parameter
     */
    void setInput(String key, String value);

    /**
     * Sets inputs for this Module from user
     */
    void setInputs(Map<String, String> inputs);

    /**
     * Checks if user has submitted any input
     *
     * @param key - key of the parameter
     * @return - if key exists
     */
    boolean hasInput(String key);

    /**
     * Clear the user inputs, prepare for new
     */
    void clearInputs();
}
```

I don't like `I` prefix for interfaces, but can't find better name, since it just encapsulates the public methods from `Module` class.

PatrykMilewski, 2018-11-24:
Exactly what I was thinking about, it looks really clear and user-friendly, let's do it

PR #110 (2018-11-26): Module refactoring
added missing JavaDocs
added `IModule` abstraction layer to `Module` class
refactoring
closes #107 

Took 52 minutes
closed 2018-11-26 by stasbar